RogueMaster/flipperzero-firmware-wPlugins #15: Unable to change default name in SAVE RAW

Eng1n33r, 2022-05-18:
**Describe the bug**
Unable to rename raw filename.

**To Reproduce**
Steps to reproduce the behavior:
1. Save some raw signal
2. Try to change defualt name
3. See error with sd-card icon (cant rename)
4. Thats it.

**Expected behavior**
Saving with changed name as it should be.

**Target**
F7

**Additional context**
In my unleashed fork I will try make temp solution for it.

Eng1n33r, 2022-05-18:
Seems to be fixed in new update allready. Or it is SD card bug. I have saved 3 signals with changed name, but 4th gave same error. But renaming existing file is allways OK. Very strange...

closed 2022-05-18 by Eng1n33r

RogueMaster, 2022-05-18:
Maybe SD bug, I tried to recreate and it renamed properly. I'm using a 128GB SD card from Brands Mart. Forgot the brand and my SD card is taped in.

Eng1n33r, 2022-05-19:
> Maybe SD bug, I tried to recreate and it renamed properly. I'm using a 128GB SD card from Brands Mart. Forgot the brand and my SD card is taped in.

Mine is also 128GB. SanDisk Ultra XC I
Fortunatly, bug is no longer appears.

commit 788848ae (2023-11-06): Create FUNDING.yml closes #15