RogueMaster/flipperzero-firmware-wPlugins #24: NFC changes in this repo makes crush if try to emulate Nice FloR-S protocol.

Eng1n33r, 2022-05-19:
**Describe the bug**
By elimination method I figured out that NFC part changes leads to hard fault if you try emulate Nice FloR-S protocol

**To Reproduce**
Steps to reproduce the behavior:
1. Open Nice FloR-S signal
2. Press button 'SEND'
3. Flipper reboots with HARD FAULT message

**Expected behavior**
No conflicts

**Target**
F7

RogueMaster, 2022-05-19:
Could you try and build https://github.com/RogueMaster/flipperzero-firmware-wPlugins/tree/ab9b667c7db5e7c27cd5dbfc20e06c87eb2c155e and let me know if it fixes it

Eng1n33r, 2022-05-20:
> Could you try and build https://github.com/RogueMaster/flipperzero-firmware-wPlugins/tree/ab9b667c7db5e7c27cd5dbfc20e06c87eb2c155e and let me know if it fixes it

I think I made same changes like in your patch commit. So it should compile correct.

r3df0xx, 2022-05-20:
But there is still a problem with nfc changes. Please check, as it leads to hard fault in runtime.

Eng1n33r, 2022-05-20:
Please check NFC also, as it makes hard fault.

RogueMaster, 2022-05-20:
Latest update to main builds revert both sets of changes, thank you for your testing

RogueMaster, 2022-05-20:
#1238 #1242 heads up these fixes made Nice FloR-S stop working per Eng1n33r

closed 2022-05-20 by RogueMaster

RogueMaster, 2022-05-20:
Hey @Eng1n33r , if either of these pulls get approved, #1238 or #1242, the main dev update will break that Nice FloR-S emulation. Any idea if it is caused by a bug in the changes, or just a change in protocol support? If Nice FloR-S support can be updated to work with the pull changes, you may want to consider if they get approved.