RogueMaster/flipperzero-firmware-wPlugins #58: BadUSB scripts no longer working in 0.60.3-7618c8ba-0617-RM

Steakumms, 2022-06-18:
After flashing to **0.60.3-7618c8ba-0617-RM**, my BadUSB scripts no longer working.  I flashed to official and they run fine, then back to 0.60.3-7618c8ba-0617-RM and no longer working.

**Steps to reproduce the behavior:**
1. Flashed to **0.60.3-7618c8ba-0617-RM**
2. BadUSB scripts no longer working, tried multiple including demo
3. Flash latest official firmware **0.60.2**
4. BadUSB working fine
5. Flash back to 0.60.3-7618c8ba-0617-RM
6. BadUSB scripts no longer working again

**Target**
BadUSB

RogueMaster, 2022-06-18:
Ok, reported upstream and reverting

closed 2022-06-18 by RogueMaster

RogueMaster, 2022-06-18:
@Steakumms please see if [new DFU 0.60.3-7618c8ba-0618-RM](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/releases/tag/0.60.3-7618c8ba-0618-RM) fixes your issue

Steakumms, 2022-06-18:
Just tested and all confirmed working, thanks to all for the quick fix