RogueMaster/flipperzero-firmware-wPlugins #70: Latest release labelled as 0.61.0 but reports as 0.62.0 on Flipper

hasmar04, 2022-06-28:
**Describe the bug**
Release  0.61.0-0627-RM is labelled as version 0.61.0, but on installation on Flipper it says it is 0.62.0

**To Reproduce**
Steps to reproduce the behavior:
1. Install 0.61.0-0627-RM
2. Hold down button on desktop to view version
3. Check version number next to build date. 

**Expected behavior**
The version of the release should be the same as the version the Flipper reports

**Additional context**
I also noticed that the release does not list the dev build version from upstream in the file name or the release notes.

RogueMaster, 2022-06-28:
Indeed, I realized that I baked in the wrong version and fixed it in the code already. Next revision will have the correct version.

RogueMaster, 2022-06-28:
As for the dev build, yeah I had to change my OS from CentOS 7 to Debian 11.3 in order to compile new scons build. Therefore, this is how they will show for now until I figure out how to make it mirror main. If you have a fix, please advise. Also will likely work on changing "unleashed" to say "dev"

closed 2022-06-28 by RogueMaster