RogueMaster/flipperzero-firmware-wPlugins #77: Games to remove

RogueMaster, 2022-07-04:
## **POLL:**
- Option 1 Raycast [Heart Emoji]
- Option 2 Game of Life [Rocket Emoji]
- Option 3 Flappy Bird [Eyes Emoji]

RogueMaster, 2022-07-04:
Heart+Rocket = 8KB

RogueMaster, 2022-07-04:
games will stay in code but removed from meta/applications.fam

UberGuidoZ, 2022-07-04:
Raycast is as close to Doom as we have... but I get it's "pointless" like game of life. But game of life COULD be good. Flappy Bird, to me, doesn't even work as you never die. If there was a challenge, I'd say keep it. But you can just go through walls. =P

ESurge, 2022-07-04:
I say whichever combo is smallest until sd card loading is ready. :)

theryex, 2022-07-04:
You could always make a second build with zero games for more features, until SD loading that is.

pingywon, 2022-07-04:
> Raycast is as close to Doom as we have... but I get it's "pointless" like game of life. But game of life COULD be good. Flappy Bird, to me, doesn't even work as you never die. If there was a challenge, I'd say keep it. But you can just go through walls. =P

Conway's Game Of Life is a hacker classic though.... :*( 

My vote is for raycast. It just doesnt do it for me. It is great but it tries to be too much on a limited platform. my 2 cents

RogueMaster, 2022-07-05:
Surprisingly, flappy really got the axe. I have axed all 3 right as it is needed with the latest updates. Can't wait for SD loading! May add Chess to test and compile and hide it...

closed 2022-07-05 by RogueMaster

RogueMaster, 2022-07-05:
> You could always make a second build with zero games for more features, until SD loading that is.

This can easily be done by compiling with a meta file without games, I have no interest in compiling a boring firmware without the games...