RogueMaster/flipperzero-firmware-wPlugins #110: plugins highlighted with ELF not seen in the plugin menu

amiseh, 2022-07-22:
Guys,

can You tell me what should I do to see all the plugins which are included in this firmware on my flipper?? I can see only few of them link:
- HID analyzer
- PicoPassReader
- TOTP
- Wifi scanner

so the plugins without ELF highlight/information before the name of the plugin... 

[image]

what to do to make rest of them visible and working?

xNymia, 2022-07-22:
The rest of these plugins are ELF binaries and are available under the ELF Loader main menu option.

amiseh, 2022-07-22:
ok, found out what was wrong. had to manually put apps folder onto SD card and just after that was able to run application with EFL loader. 

sorry. it was my mistake.

closed 2022-07-22 by amiseh