RogueMaster/flipperzero-firmware-wPlugins #111: The elf loader is not working properly

Hyralc, 2022-07-23:
**Describe the bug**
When y use the elf loader, i always get an error:

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on...
2. Press button '....'
3. Wait for the moon phase
4. It burns

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
Add debug logs

**Target**
Specify the target

**Additional context**
Add any other context about the problem here.

RogueMaster, 2022-07-23:
Did you install using readme instructions and the archive app?

Hyralc, 2022-07-23:
i followed the instructions here: https://flipper.pingywon.com/
i downloaded the whole archive containing everything and put it in the right place, and the files thas goes to subgz/assets too

Hyralc, 2022-07-23:
maybe the resources there are outdated ?

RogueMaster, 2022-07-23:
That's not my page and I don't suggest installing with DFU. Please use the readme

Hyralc, 2022-07-23:
ok i corrected it, thanks and sorry

closed 2022-07-23 by Hyralc