RogueMaster/flipperzero-firmware-wPlugins #117: Restore the emotion-dependent animations

christiankuttke, 2022-07-25:
### Description of the feature you're suggesting.

I would like to have an option to make the dolphin not cycle through all animations, but instead play the animations according to its mood.

In "All Changes/Features" in this repo's README.md, the relevant point is "Animations: Idle animations will show all animations regardless of level and butthurt [Thanks to qqMajiKpp]", but I couldn't find any more information about this.

KairuByte, 2022-07-25:
A setting to choose default/all would indeed be nice.

RogueMaster, 2022-07-25:
There is a Manifest file in the assets/resources/dolphin directory to help. Just replace it with the one from OFW and you should regain your mood dependent animations.

closed 2022-07-25 by RogueMaster

sdaitzman, 2022-11-02:
I've been restoring the original animation manifest with each update successfully. For the most recent update as of now, when I restored the stock manifest file, I was only able to see two animations (flipper painting, and flipper watching TV, none of the others). Any ideas why this may have changed?

RogueMaster, 2022-11-02:
you probably aren't installing from SD...

sdaitzman, 2022-11-02:
I ran the update by unzipping and copying the folder to the SD card directly, is that what you mean?

On further investigation, some (all?) of the animation folders in the flipper folder were empty. I restored their contents from a backup and the manifest file from that same backup, and was able to restore the same animations as before.