RogueMaster/flipperzero-firmware-wPlugins #118: Name change and FW building

ctrlcmdshft, 2022-07-25:
### Describe the enhancement you're suggesting.

I would like to change the name of my flipper and i know the file is firmware/targets/f7/furi_hal/furi_hal_version.c and to search for CH0NG but what do you change in this code?  And when building FW via Docker how does one know it compiled correctly?

```
const char* furi_hal_version_get_name_ptr() {
    return *furi_hal_version.name == 0x00 ? NULL : furi_hal_version.name;
    // return "N4V1RUK4";
    // return "N00BY";
    // return "CH33CH";
    // return "CH0NG";
}
```

Thanks

RogueMaster, 2022-07-25:
Change to : 

const char* furi_hal_version_get_name_ptr() {
    // return *furi_hal_version.name == 0x00 ? NULL : furi_hal_version.name;
    // return "N4V1RUK4";
    // return "N00BY";
    // return "CH33CH";
    return "CH0NG";
}

Where CH0NG is the name you are changing to. You can confirm it worked by your changed flipper name when you install.

closed 2022-07-25 by RogueMaster

ctrlcmdshft, 2022-07-26:
builds and installs fine but isnt changing name