RogueMaster/flipperzero-firmware-wPlugins #122: [NFC] Mifare-Classic can be cracked and sectors read, but unable to emulate it correctly

simkard69, 2022-07-28:
### Describe the bug.

After decrypting A and B keys for a Mifare-Classic, everything went well and saved.
At time of emulating, it seems that reader refuses to accept the emulated tag.

### Reproduction

Crack any NFC Mifare-Classic tag using brute-force (dictionnary) attack and then try to emulate it in front of the reader > doesn't work.

### Target

NFC

### Logs

```Text
Really needed ?
```

RogueMaster, 2022-07-28:
Please submit this upstream to main repo; I can't help with this

closed 2022-07-28 by RogueMaster

simkard69, 2022-07-28:
> Please submit this upstream to main repo; I can't help with this

Already existing on MAIN : [NFC Emulation of previously saved Mifare classic 1k card does not work. #1468](https://github.com/flipperdevices/flipperzero-firmware/issues/1468)