RogueMaster/flipperzero-firmware-wPlugins #123: [NFC] After saving tag, ability to write the dump contents to a fresh tag (with sector 0/UID unlocked)

simkard69, 2022-07-28:
### Description of the feature you're suggesting.

After a successful read of a protected tag (with A and B keys), adding a menu option to write the dump contents (with protected keys, etc ...) to a brand new and fresh tag.
As NFC tag emulation tends to have problems with protected tags (A and B keys), it might be a good option here.

RogueMaster, 2022-07-28:
This would be a cool option, please request on main to see if it can get some attention.

closed 2022-07-28 by RogueMaster

simkard69, 2022-07-28:
> This would be a cool option, please request on main to see if it can get some attention.

Seems like already existing on MAIN : [Support writing NFC tags #1353](https://github.com/flipperdevices/flipperzero-firmware/issues/1353)