RogueMaster/flipperzero-firmware-wPlugins #126: Failed Compile

AFstache, 2022-07-28:
I've tried compiling manually and after doing the git clone --recursive, I run./fbt resources icons with no issues, but when I run ./fbt --with-updater updater_package I get "Failed building firmware/targets/f7/api_symbols.csv: SDK version is not finalized, please review changes and re-run operation" am I missing a step?

edrethardo, 2022-07-28:
looks like the main branch is broken rn as I'm having the same issue.
I tried my stuff on a branch forked from the latest release tag instead and it worked like a charm.

RogueMaster, 2022-07-28:
Please compile with code from 07-25. Like it says in the readme and 7/27 release, some stuff is broken right now.

closed 2022-07-28 by RogueMaster