RogueMaster/flipperzero-firmware-wPlugins #130: Missing `WIFI (MARAUDER)`

ThornWalli, 2022-07-29:
Where can I find `WIFI (MARAUDER)` in the latest release?

RogueMaster, 2022-07-29:
it needs to be compiled in at the moment, I will do some space tests and likely build it into next release

closed 2022-07-29 by RogueMaster

Silentscreech5, 2022-07-29:
ok cool had me worried

sdaitzman, 2022-09-21:
Hi, I'm having an issue with WiFi Marauder, and saw this as a potential duplicate. Should it now appear in the menus? I can't find the control interface pictured in this video: https://youtu.be/Deh5NBr0e_A?t=71

In fact, I see no "plugins" menu at all, and no marauder details in the browser except for Applications → GPIO → [ESP32] WiFi Marauder. That application has a very different interface than the one pictured in that video. Is that simply because the interface has been updated?