RogueMaster/flipperzero-firmware-wPlugins #131: Clock app should follow system settings for volume

danielskowronski, 2022-07-29:
### Describe the enhancement you're suggesting.

Currently, clock app emits sounds even when system volume setting is set to 0%. 

If changed to follow system settings, it would greatly improve stealth mode, since it's easy to press enter button, which starts the stopwatch - on every restart of the app, it reverts to default sound settings (Mario).

Behaviour present on 0.63.2-0728-RM-Tv2 and 0.62.3-0725-RM

RogueMaster, 2022-07-29:
Yes, you are absolutely correct; it should. Would you like to look into assigning those settings? @danielskowronski

danielskowronski, 2022-07-29:
I can try :)

Ganapati, 2022-08-17:
PR with a patch : https://github.com/RogueMaster/flipperzero-firmware-wPlugins/issues/131

RogueMaster, 2022-08-17:
Fixed by Ganapati

closed 2022-08-17 by RogueMaster