RogueMaster/flipperzero-firmware-wPlugins #147: MouseJack is crashing

Einstein2150, 2022-08-01:
### Describe the bug.

MouseJack is crashing

### Reproduction

Start Mouse Jack an the flipper, select a file and then it crashes with "HardFault"

### Target

flipperzero 0.63.3-0801

### Logs

```Text
none
```

### Anything else?

NRF Scanner is working without a problem

RogueMaster, 2022-08-01:
I can't test this too well so I can't help with fixing it

RogueMaster, 2022-08-02:
@mothball187 Would you have time to test this and see if it is a concern? Thank you for your development!

mothball187, 2022-08-04:
@Einstein2150 which file are you selecting that causes the crash?

Einstein2150, 2022-08-04:
> @Einstein2150 which file are you selecting that causes the crash?

I added a manual addresses.txt with one single rate and address value. Maybe there is an error in the interpretation or I missed the correct format of the string in the file.

I'm getting a Logitech R400 Presenter in the next days and capture the real traffic. Maybe then I can inspect deeper what the problem is ...

Einstein2150, 2022-08-05:
Ok. I select the addresses.txt - its containing the mac and the data-rate i.e. 382493c007,2 (I sniffe the device with a crazy radio-pa - not sure using the normal hex order or the reversed one 07c0932438 ...)

After selecting the ducky-script the screen stays blank

mothball187, 2022-08-05:
The screen will stay blank until the ducky script is finished executing. I did not implement any kind of loading/progress screen in the app. If the keystrokes are not being injected, I would try reversing the address bytes in the file. Also be sure you are using the correct newline character ("\n", not "\r\n") in the addresses.txt file

closed 2022-08-07 by RogueMaster

Einstein2150, 2022-08-07:
@mothball187 works with Logitech R400! It needs reverse byte order and 2 mBit speed. All other combinations don't work. I will contribute with a better documentation of the function here to help the others 👍

mothball187, 2022-08-07:
@Einstein2150 excellent! glad to see you got it working with an external capture source.