RogueMaster/flipperzero-firmware-wPlugins #149: BadUSB: DUCKY_LANG not working

gilfy, 2022-08-01:
### Describe the bug.

I am adding DUCKY_LANG DE to the first or the second raw and the payload is still in default

### Reproduction

Switch you keyboard setting to DE and execute the payload via Flipper

RogueMaster, 2022-08-01:
Please let me know as you test how you come to a solution

RogueMaster, 2022-08-01:
Try it exclusively on line 1 and let me know if it fixes it

gilfy, 2022-08-02:
As I already mentioned exclusively line 1 or line 2 does not work unfortunally :(

carnyccc, 2022-08-02:
Hi I had the same problem and was able to solve it by changing the script file in Notepad++ from Windows CR/LF to Unix LF.

gilfy, 2022-08-02:
Ya @carnyccc but this is not the goal to set smth by your own. Imagine you have a PC of your friend etc. you cannot setup first his PC :D

RogueMaster, 2022-08-02:
> Hi I had the same problem and was able to solve it by changing the script file in Notepad++ from Windows CR/LF to Unix LF.

Thank you for the solution 😄

closed 2022-08-02 by RogueMaster