RogueMaster/flipperzero-firmware-wPlugins #150: Hardfault when read new mail

galactricks, 2022-08-01:
### Describe the bug.

When I click left on the new mail for pass to the lvl 7  flipper reboot and hardfault 

### Reproduction

1 press left for read new mail
2 reboot and hardfault

### Target

0.63.3-0801-RM-LVLS

RogueMaster, 2022-08-01:
Hard fault and it boots up again or does it reboot infinitely?

galactricks, 2022-08-01:
First time after the update he reboot indefinitely 
So I'm turn it to dfu mode and put all file on the update folder now on read mail he just reboot crash but after I can use it is boot up again

RogueMaster, 2022-08-01:
Ok I'll try to force my flipper to level and test out if I can recreate. Did you do the update with SD card? You need the updated dolphin/manifest.txt file for levels over 3 but I also haven't tested leveling on higher than 3

galactricks, 2022-08-01:
I did not modify the manifest it must probably come from here
And yes I update with SD
before the update I was close to level 3 and after the update I went to lvl 6 with the same amount of xp and I received the mail for lvl 7

galactricks, 2022-08-01:
The manifest is good

galactricks, 2022-08-01:
https://user-images.githubusercontent.com/9158423/182255414-0df81e46-0ad9-401f-bf4e-750a831b85ee.mp4

RogueMaster, 2022-08-02:
Could you compile and test current build to see if new mail causes fault?

RogueMaster, 2022-08-02:
Confirmed this is fixed

closed 2022-08-02 by RogueMaster

galactricks, 2022-08-02:
that good after your morning update thanks