RogueMaster/flipperzero-firmware-wPlugins #155: Button bounce

port513, 2022-08-02:
### Describe the bug.

All buttons do bounce from time to time. This should be fixed in hardware but now we need to have it done by software instead.

It applies to the direction button, OK and back button.
It sometimes register that the button is pressed more than twice that is annoying. 

### Reproduction

Just try any menu, unlock code etc and you will see this behavior. Alot of the times I'm unable to unlock the device due to these bounces. 

### Target

Latest RougeMaster FW

RogueMaster, 2022-08-02:
I found similar issues especially entering the password. Not sure what changes can be causing it or what can fix it.

RogueMaster, 2022-08-03:
Does this till present a great issue in latest build? The logging was somewhat updated

port513, 2022-08-03:
> Does this till present a great issue in latest build? The logging was somewhat updated

Had this problem with stock firmware and other custom firmware etc. So not worse now than before.

RogueMaster, 2022-08-04:
Please review after todays release later today and let us know if this is still an issue

closed 2022-08-04 by RogueMaster