RogueMaster/flipperzero-firmware-wPlugins #157: Sub-ghz auto read crash

FrozenGhostx, 2022-08-02:
### Describe the bug.

When you navigate to sub ghz, select auto read and select in options "detect raw", scan on 433.88MHz  and AM670 Emulation and search for keys, it finds them and you can select them. If you try to click on one of the found keys, flipper crashes and can't boot up anymore. Goes in DFU, you just can re-flash it.

### Reproduction

SubGHZ
Read
Detect RAW
Found Keys
Select them
Crash
Destroyed (just kiddin'. DFU.)

FrozenGhostx, 2022-08-02:
EDIT: Re-Tried, as soon as i try to replay the detected raw, it fully crashes and can't boot up anymore.

Einstein2150, 2022-08-02:
Tried at 434.42 AM650 Raw Detection on - no Problem while scanning, saving and replaying here ... 🤷🏻‍♂️

FrozenGhostx, 2022-08-02:
> Tried at 434.42 AM650 Raw Detection on - no Problem while scanning, saving and replaying here ... 🤷🏻‍♂️

Tried on 433.88 and 433.92? These both crashes. Already re-flashed it 3 times to be sure.

perspecdev, 2022-08-02:
There's a bug where if you click the Send button before saving the signal, it will crash.

RogueMaster, 2022-08-03:
This option was fixed on #160

closed 2022-08-03 by RogueMaster