RogueMaster/flipperzero-firmware-wPlugins #158: Desktop settings being loaded constantly, plus extra logging

perspecdev, 2022-08-02:
### Describe the bug.

Looking at `log`, it seems the desktop settings are being loaded constantly.  This line is in the logs every 0.5-1 second:
`Loading "/int/.desktop.settings"`

Additionally, while in the subghz app, these lines are rapidly logged over and over:
```Checking if frequency is valid
Using extended frequencies at own risk```
There should not be any unnecessary logging like this.  This is very likely to slow down the Flipper and make subghz feel sluggish and even miss signals.

### Reproduction

Connect to Flipper via ssh and type `log`.  Browse in any app to see that the desktop settings are being constantly loaded.  Attempt to capture a signal in the subghz app to see the extra lines being written to the log.

perspecdev, 2022-08-02:
There's also likely a significant impact from reading the extended range file on every tick when reading subghz.  I posted in the Unleasedhed Discord about it, since that fork appears to be the origin of this:
https://discord.com/channels/937479784148115456/937489970007003166/1004031760968974427

RogueMaster, 2022-08-02:
@tkerby Would you be able to see where we can put the extended check? @perspecdev, @tkerby had added that for the extended ranges

RogueMaster, 2022-08-02:
I removed the extra logging

tkerby, 2022-08-02:
Extra logging was mine too. It used to only load the checks once when you entered the subghz app and it logged once for each frequency on the list but didn't log when running. Same went for range checking. I think the latest official updates are somehow causing those routines to be run on every rf hop. It'll be both the extend range and also the region unlock being read.

Is there a mechanism in FZ to cache settings as some sort of system global? The ideal would be to have a settings menu option for subghz that we can select frequency ranges, region unlock, choose which frequencies go into hopping etc

RogueMaster, 2022-08-02:
Loading "/int/.desktop.settings" happens to test for game mode; anyone have a suggestion for alternative location for checking and storing this? 

As for alternative location for checking and storing the subghz settings, I think Nano from the Eng1n33r build is also looking into an alternative location. Their build helps a lot for SubGHz stuff since I'm limited on my knowledge with that.

perspecdev, 2022-08-02:
> Extra logging was mine too. It used to only load the checks once when you entered the subghz app and it logged once for each frequency on the list but didn't log when running. Same went for range [...]

There may be a better solution, but one thing you could do is put it inside `subghz_tx_start`.  File IO is already happening there.  Then you would need to pass it as a bool parameter on down to `subghz_tx`, and then as a bool parameter from there to `furi_hal_subghz_is_frequency_valid`.

perspecdev, 2022-08-02:
> Loading "/int/.desktop.settings" happens to test for game mode; anyone have a suggestion for alternative location for checking and storing this?
>
> As for alternative location for checking and storing [...]

I haven't looked at the game mode or desktop code.  Maybe just do the check when the desktop scene starts and when the menu loads?

RogueMaster, 2022-08-04:
Ok, I moved the settings out. Unlock still seems to operate. Please review new location and re-open if it still has issues

closed 2022-08-04 by RogueMaster

RogueMaster, 2022-08-04:
that settings move didn't work as expected so the settings are in the same place...