RogueMaster/flipperzero-firmware-wPlugins #161: SubFrequenzy - Read / reply RAW crashes flipper

thespawnat, 2022-08-03:
### Describe the bug.

I've installed the latest FW-Update today morning. 

I'm running the SubGhz in Read with Settings: Hopping = ON ; Modulation = AM650 ; Detect Raw = ON.

the sniffer works and creates the entries in the list. When I open an entry and emulate, the flipper screen goes off and is unresponsive. with "left + back" (soft reset) it gets back to life and provides an error message about hard fault

### Reproduction

install latest FW
open SubGhz
open Read
set to "hopping = true" 
set to "Modulation = AM650"
set to "Detect RAW = true"
start sniffing
once entry appears -> open it and press "emulate"
-> flipper crashes

RogueMaster, 2022-08-03:
Will emulating after a flipper restart still cause a bug/crash?

thespawnat, 2022-08-03:
If the file is saved and opened via subghz/saved I can emulate it properly.

If I save it and directly try to emulate from 'read-mode' it crashes too (blank screen, no led, no response)

RogueMaster, 2022-08-03:
@perspecdev Could emulate from read-mode be fixed or disabled? Thank you for your help!

tim427, 2022-08-03:
I can indeed confirm that the flipper crashes from "read" and directly "replay".

Saving and "replay" from "saved" works just fine.

perspecdev, 2022-08-03:
Thanks for this.  Is there a new "emulate" button?  As far as I know, there was only "send" button, and I disabled it with the latest PR, but maybe I'm not on the latest code.  I'll update everything in a bit and take a look.

galactricks, 2022-08-03:
Hello I have the same issue 🙃

av4k1r, 2022-08-03:
After captured with the "RAW" option,when "SEND" is pressed, it restarts and "Hard fault" is displayed

av4k1r, 2022-08-04:
After capturing with "RAW" option and pressing the "SEND", the Flipper freeze. I had to press "left & back" to reboot, it displays "NULL pointer dereference"

Mika73, 2022-08-04:
Just updated to latest firmware. Now when trying to read wireless carkey Flipper crash and reboot everytime.

thespawnat, 2022-08-04:
installed latest update today morning. Behavior is the same as with FW from yesterday with "Send" crashes FZ in "read"-mode (nullpointer reference). 
 when I try "read" with RAW capture enabled I'm unable to detect my car key on 868,35. if I make a "read raw" on the same Frequenz, I'm able to capture the signals and send them directly from "read raw" without reboot / freezes

RogueMaster, 2022-08-05:
This should be fixed on last release

closed 2022-08-05 by RogueMaster