RogueMaster/flipperzero-firmware-wPlugins #162: Add GFSK to SubGHZ

Mik13, 2022-08-03:
### Describe the enhancement you're suggesting.

There should be a possibility to read and save chats done on the cli (subghz chat).

I already found out it uses the GFSK modulation which furi knows, but the subghz application is missing.

### Anything else?

I tried my luck, but it does not work:
https://github.com/RogueMaster/flipperzero-firmware-wPlugins/commit/f7c128cb5861b4b35bbe3c1355d359015d482740

When using Read RAW with GFSK and the correct frequency. the flipper receives the message (when using another flipper and sending a message there), spl goes to 2 and the bar goes up, but never down as it should. Stopping recording and sending does nothing (aka, the flipper in chat does not receive anything).

Could you point me at the correct direction about what I am missing? Thank you in advance!

Mik13, 2022-08-03:
Sadly I can't work on this issue very much because I only own one flipper and no SDR or other device to send chat messages from to my flipper.

RogueMaster, 2022-08-04:
I can't help you with this request; perhaps submit it upstream to see if they can help

closed 2022-08-04 by RogueMaster