RogueMaster/flipperzero-firmware-wPlugins #163: DUCKY_LANG not Working

J0s3f, 2022-08-03:
### Describe the bug.

I want to run a script and have it working on Windows with German keyboard layout.
I tried with that script: [Hacker_Typer.txt](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/9251441/Hacker_Typer.txt)
It's not working. The key codes are still sent with US layout.
Am I doing something wrong, or is there a bug?

How could I best debug this? There is a FURI_LOG_D statement in the code. Is there an easy way to see this log? https://github.com/RogueMaster/flipperzero-firmware-wPlugins/blob/10fd353e882a0efcc5c0b9ea37547364276cfe8e/applications/bad_usb/bad_usb_script.c#L338 

### Reproduction

1. put attached script on flipper storage
2. set windows keyboard to German
3. run script
4. get error message about malformed URL

RogueMaster, 2022-08-03:
Please see if you have the same issues with the original from @v1nc https://github.com/v1nc/flipperzero-firmware

carnyccc, 2022-08-03:
Hi J0s3f,
I had the same problem. The solution is to check the Line Feed of your Hacker_Typer.txt. You need to use LF not CR/LF. If you use an editor on Windows, it changes the Linefeed to Windows style CR/LF.
https://github.com/RogueMaster/flipperzero-firmware-wPlugins/issues/149#event-7111569078

After Change this via Notepad++ and put DUCKY_LANG DE to the first line of my script, it works for me.
I hope this helps.

carnyccc, 2022-08-03:
Hey me again,

on the readme.md file of https://github.com/v1nc/flipperzero-firmware they write the following:

_If you are writing payloads on windows you need to make sure EOL Conversion is set to LF Unix in your text editor and not windows CR LF. If you do not it will fail to run the payload._

J0s3f, 2022-08-03:
It works for me with the original firmware and now also works in this fork.
Only way I can explain this is, that I made some mistake flashing the firmware the first time.
Sorry for bothering you.

closed 2022-08-03 by J0s3f

gilfy, 2022-08-05:
It is working now with the new release. Before you run your script you can click left to configuration and choose the keyboard layout in beforehand