RogueMaster/flipperzero-firmware-wPlugins #173: crash loop after latest update

zviratko, 2022-08-04:
### Describe the bug.

Today I got a boot loop / crash after flashing
0.63.4-0803-RM-BU2

I don't know that caused it (it crashed right away after boot). After reflashing from DFU mode to latest DEV it still continued crashing but in a different manner, it took a few seconds on the home screen to crash, so I was able to do a local update with the previous release using File Manager which fixed it.

I don't know what caused it, I did a by-the-book update by unzipping the release on the sd card and doing a local update which should include all needed resources etc., so it would be a bit surprising if it only happened to me.

Also very surprising that older binary releases are no longer available here on GitHub? I had to recompile with the latest release hash, not something an ordinary user should be expected to do...

### Reproduction

Flash latest release - 0.63.4-0803-RM-BU2

zviratko, 2022-08-04:
I should add that I also tried flashing the head of the unleashed branch (up to 713a76ebe78c2db7948f7d0a000c4c2d8533e63f) which didn't fix it.

RogueMaster, 2022-08-04:
Crash boot sequence after update recently was caused by a DFU update without updated manifest for dolphin files. Did you do SD card update the first time or update through qFlipper?

zviratko, 2022-08-04:
I update every day to your new release using the sd card / file manager / update.fuf. Not sure if that's the same as updating via qFlipper (but I assumed it is). I did the same today. I already had the new dolphin levels firmware and everything worked correctly, it was unly this particular update that caused this.

zviratko, 2022-08-05:
Today's update went without a glitch (sd update.fuf method).

zviratko, 2022-08-05:
Rejoiced too soon - after changing ignore_default_tx_range to true it started crash looping again.
Probably not the same bug as yesterday, but still weird.

zviratko, 2022-08-05:
but it doesn't crash loop if SD card is not inserted, so I can remove it, reboot, insert it and reflash... this is really weird.