RogueMaster/flipperzero-firmware-wPlugins #175: Frequncy Analyzer Interference

ResistanceIsUseless, 2022-08-04:
### Describe the bug.

When running the frequency analyzer it always picks up 468.000Mhz or 467.989Mhz right away. When troubleshooting and trying to see if this is a legitimate signal or not, nothing shows up in a raw capture. Using my hackrf and seeing if there is anything local that could be causing the signal i see nothing. On discord I brought this up and it sounds like other users are seeing the same behavior. I currently have all frequencies unlocked but i was seeing this happen before i did that. I tried upgrading to 0.63.4-0804-RM-NOFAP from 0.63.3-0802-RM-LVLS and it still is seeing this signal.

My guess is that there is either interference coming from the flipper itself or a bug in the Frequency Analyzer app. Maybe adding a sensitivity setting or block out 468Mhz unless it's a certain db?

### Reproduction

Switch on and go to SubGhz
Select Frequency Analyzer

RogueMaster, 2022-08-04:
I don't see this issue at all, you mast have devices with that frequency

ResistanceIsUseless, 2022-08-04:
I guess i must have an implant because no matter where i am even miles away from my home i still see this behavior.

This is the most "sucks to be you" response. The least you could have done was left it open to see if other people have the issue or changed it to enhancement. Regardless of who's right, it's annoying to be stuck on a frequency.

RogueMaster, 2022-08-04:
Check if you have the same issue on main FW, they have people that know more about the subghz implementation...

RogueMaster, 2022-08-04:
Also, I may roll it back of a PR of some changes 1501

ResistanceIsUseless, 2022-08-04:
Just loaded main FW 0.63.3-RC and it's fine. No issues, works fine and picks up a test from a fan remote.

UberGuidoZ, 2022-08-05:
@ResistanceIsUseless try Unleashed and see. Since RM is based off it, that's the next logical step to see where this is happening.

RogueMaster, 2022-08-05:
Seems like its an issue...

Einstein2150, 2022-08-05:
Same issue here ...

sliceratwork, 2022-08-05:
I don't think it's a bug in the frequency analyzer. It could be internal interference.

Can anyone have a look in the spectrum analyzer and see if they get a small peak around  468MHz?

Have a look at the attached image.

[image]

Einstein2150, 2022-08-05:
> I don't think it's a bug in the frequency analyzer. It could be internal interference.
> 
> Can anyone have a look in the spectrum analyzer and see if they get a small peak around 468MHz?
> 
> Have a look at the attached image.
> 
> [image]

I can confirm this too (little peak, very small) but in official latest dev-stock this issue is not present 🤷🏻‍♂️

Kuronons, 2022-08-05:
[image]
Damn... I have noise on all frequencies

IceAgesz, 2022-08-05:
It occurs to me every time I use Frequency Analyzer But I don't have any implant at all. When I use an OFW It works just fine. (I also did Unleash )

sliceratwork, 2022-08-05:
@Kuronons do you have the flipper connected to the computer via a USB cable? Asking because I don't know how you took that "screenshot" :).

If so, that would explain all the noise.

Kuronons, 2022-08-05:
> @Kuronons do you have the flipper connected to the computer via a USB cable? Asking because I don't know how you took that "screenshot" :).
> 
> If so, that would explain all the noise.

Same if connected with usb or not... I lowered the sensibility (not sure how to call that XD) and I have still lil noises here an there... but I don't have the visible peak at 468 like you have.

[image]
[image]

RogueMaster, 2022-08-05:
Fixed for latest "TAMA" release

closed 2022-08-05 by RogueMaster