RogueMaster/flipperzero-firmware-wPlugins #176: Release documentation specificity

bbbenji, 2022-08-04:
### Description of the feature you're suggesting.

I appreciate this project as well as the detailed release documentation. However, it makes finding the changes very hard for each release, which occurs quite frequent. I'd like to suggest providing only changes in the release information and expanding the readme with the stable changes that are currently in the release info.

I'd be happy to help rewrite the readme and submit a PR if you are willing to accept my suggestion and only provide actual changes in the release info. Please let me know.

RogueMaster, 2022-08-04:
Release notes of current build are purposefully the whole read me. This wont change. I'm open to any rewrites you suggest.

bbbenji, 2022-08-04:
Why duplicate the release notes every release? It's all mostly static content that should be contained within the readme?

closed 2022-08-05 by RogueMaster

bbbenji, 2022-08-22:
Thanks for cleaning up the release notes and only outlining the changes. They look great now! And everything is in the README. Thanks!