RogueMaster/flipperzero-firmware-wPlugins #180: Flipper still crash with TAMA update. Null pointer derefence.

Mika73, 2022-08-05:
### Describe the bug.

Flipper does still crash with latest Tama update. When try read Sub-Ghz. 

### Reproduction

Still crash with Sub-Ghz read.

Mika73, 2022-08-05:
[Fiat.sub.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/9267086/Fiat.sub.zip)

Send/Emulate this totally crash Flipper. I managed read file with hopping mode.

Mika73, 2022-08-05:
Also downloaded new setting_user file but it didnt help crash

thespawnat, 2022-08-05:
I'm able to capture signals in "read mode" with "detect raw = on" on 433,92 and can send them directly without crash (no saving, direct send). maybe I can test with 300/312 in the next days to check that frequencies too.

I noticed an issue with 868 MHz in the new Read-Mode (separate issue created), as my car-key-signals are only detected in explicit raw-mode (read-raw) . sending from read-raw results didn't cause a reboot for me, even on the last releases, only in the new "read".

zviratko, 2022-08-05:
Also crashing on TAMA. You can "workaround" this by rebooting without the SD card and then inserting it. It will then work fine, somehow. I reformatted my SD card and did a clean install, still crashes, so it's a broken release...

RogueMaster, 2022-08-05:
@zviratko you may be missing the assets for SD from the release. Specifically the dolphin/manifest.txt file or subghz/assets/extended_range.txt

RogueMaster, 2022-08-05:
@Mika73 I can recreate the crash but only with the file you sent which seems to be specific to a special protocol Keeloq 64bit at 433.92AM. will check further

zviratko, 2022-08-05:
Nope, those are all here. I did a clean install and all resources are present.

credek, 2022-08-05:
My flipper also have this problem, crash and rebooting when i want to read a signal.

RogueMaster, 2022-08-05:
@credek please provide more information to recreate. Please also see if your problem occurs on the unleashed build. @Mika73 your file also crashes unleashed. Must have something to do with the keeloq updates but unsure

credek, 2022-08-05:
@RogueMaster I trying to read my key fob with 'Read' option. When I click a button on my remote Flipper rebooting and i get this:
[image]

RogueMaster, 2022-08-05:
Please check if unleashed release gives you the same issue

credek, 2022-08-05:
@RogueMaster no issue on unleashed and 0.63.4-0805-RM-NOFAP release, just on this last one.

RogueMaster, 2022-08-05:
@credek Please take picture of the last page of your about; it should be the same one that states "0.63.4"

credek, 2022-08-05:
@RogueMaster https://streamable.com/6sb925
Everytime on the latest release :(
[image]

sysadminsh, 2022-08-05:
I can confirm this. Keeloq definitely crashing Flipper. This issue not happens on clean "Unleashed" firmware.

juda73, 2022-08-06:
Same crash with SCANAP in  WIFI (MARAUDER)

zviratko, 2022-08-06:
I realize that I'm probably facing a different (at least in flavor) issue, but I want to note that even if I boot without an SD card and then reinsert it, flipper still crashes later (few minutes, few hours, doing nothing it just reboots and crash loops with NULL pointer deference) also just noticed that it actually boots and crashes only after about 5 seconds on Home Screen.

RogueMaster, 2022-08-06:
Try now and re-open if you still have issues

closed 2022-08-06 by RogueMaster