RogueMaster/flipperzero-firmware-wPlugins #182: Sub-GHz - Read Mode with RAW unable to capture 868 MHz

thespawnat, 2022-08-05:
### Describe the bug.

when running READ-Mode with "Detect RAW = ON" and "Frequency = 868.35" no signal is shown when pressing the button my car-key. 

in RAW-Capture mode with "Frequency = 868.35" it shows the Signals and can capture / store on the SD

looks like not all frequencies are affected, 433,92 works fine

### Reproduction

1. open Sub-Ghz
2. open Read
3. Config:
4. Frequency: 868.35
5. Detect Raw: On
6. send signals from 868.35 Device

### Target

latest (flashed 10 min ago)

### Anything else?

with the first release, where the new "Read Mode with detect RAW" was integrated, I was able to capture 868.35 MHz (files on my SD with timestamp 03.08.22 11:49)
I have one RF-Remote running on 433,92, where the Files are captured properly (Princeton 3D0001 and a few Raw captures

RogueMaster, 2022-08-05:
Could you check if the original FW dev build gives the same issue? Please advise

RogueMaster, 2022-08-06:
Try now and re-open if you still have issues

closed 2022-08-06 by RogueMaster