RogueMaster/flipperzero-firmware-wPlugins #197: SubGHz outside default range... Å

port513, 2022-08-07:
### Describe the bug.

Not sure if this is a bug or that some settings moved...

Have extended range activated and ignore default TX region activated.
For the last three or so FW this file does not work, claims that range is outside default range, why is it outside default range?

https://raw.githubusercontent.com/UberGuidoZ/Flipper/945362d3a8058f082abd7e6dbe57864d95d363f8/Sub-GHz/Restaurant_Pagers/LRS_Pagers/BruteForceRest.sub

### Reproduction

1. Switch on
2. Go to SubGHz
3. Load the file https://raw.githubusercontent.com/UberGuidoZ/Flipper/945362d3a8058f082abd7e6dbe57864d95d363f8/Sub-GHz/Restaurant_Pagers/LRS_Pagers/BruteForceRest.sub
4. It will show that this is outside the range

### Target

Latest RougeMaster FW

### Anything else?

Everything for extended range is activated and all frequencies are in setting_user file

Worked for 3 before the last 3 firmwares

RogueMaster, 2022-08-07:
Ok I tested and can confirm something is up with it. Will review

RogueMaster, 2022-08-08:
Fixed on commit 28317ca710c1cd31e2ba4a15d10b21740071824a

closed 2022-08-08 by RogueMaster

port513, 2022-08-12:
This bug is back again since today

marcstae, 2022-12-06:
I have the same problem