RogueMaster/flipperzero-firmware-wPlugins #204: Constant Locked state if pin is set and reverting to OFW

amec0e, 2022-08-08:
### Describe the bug.

I noticed if the pin is set but not locked and you go from RM 806-807 (This happened in some earlier versions too) to the Official firmware (release) the flipper will enter a constant screen locked state and you cannot escape it, the solution is to erase the device.

When comparing RM `flipperzero-firmware/dev/applications/desktop/helpers/pin_lock.c` to the official DEV and Unleashed FW I noticed `#include <cli/cli.h>` is still present in the RM file above. I'm not a coder and so I am unsure if this has anything to do with this issue but I thought I'd point it out nonetheless.

### Reproduction

1. Install the Official Stable FW
2. Set the pin in OFW.
3. Flash Unleashed FM and check you still have the Option when pressing up "Lock with pin"
4. Flash OFW stable again and you will still be able to confirm "Lock with pin" is available (and still set)
5. Now flash RM 806-807 (probably even 808) and noticed when pressing Up it now shows "Set pin"
6. Set the Pin in RM
7. Now flash back to OFW stable and you will be in a constant screen locked state and unable to escape

Steps 3 and 4 are just to demonstrate that the PIN is carried over from OFW to Unleashed with no issues.

### Target

RM-806-807

### Logs

```Text
N/A
```

RogueMaster, 2022-08-08:
With PIN set on latest RM, I downgrade to main dev and regression unset my pin. Everything else works fine.

I re-set my pin and upgraded back to RM, again need to re-configure PIN but everything else works fine.

Aside from the PIN getting wiped which is a bug from main FW, everything is working with no bugs. 

Please advise how to re-create your issue.

RogueMaster, 2022-08-08:
I have removed the excess code line thank you, don't think it made a difference though.

RogueMaster, 2022-08-08:
oddly that cli line is needed... trying to trace where its being added for main...

amec0e, 2022-08-08:
My apologies there is a Vital step i missed which is you Must Lock with Pin once and then unlock and flash RM firmware, i can successfully reproduce this on RM-807 on the OFW Dev Release