RogueMaster/flipperzero-firmware-wPlugins #207: After running Marauder you have to hard boot FZ

nastea1, 2022-08-08:
### Describe the bug.

After you run the Marauder application for say 5 minutes or so, sometimes less. When you try and operate the FZ it states "No databases found on SD card"

### Reproduction

Turn on FZ
Run Mauarder
ScanAP
List -a
Select -a (pick one)
Run DeAuth Attack for 5 minutes or so
Exit to Main Screen
"No databases found on SD Card"

nastea1, 2022-08-08:
[image]

nastea1, 2022-08-08:
IT can be resolved by hard booting your FZ. However you have to hold Left and Back every time after you use Marauder.

RogueMaster, 2022-08-08:
I'm not sure how much of this is the FZ or possibly needing a new SD card... it could be possible due to power draw or something it makes the SD stop working... not sure how to resolve... perhaps inquire with main if using the dev board for a while with CFW it could cause the SD issue?

nastea1, 2022-08-08:
Yeah I feel you, I wasn't sure where to start. You can definitely replicate it though. Not sure If I can pull more logs off it to help or not. It's more of an annoyance, because as soon as you hard boot it, it's fine. I'm using a high end SanDisk SD Card. In qFlipper, when you're connected and this happens the firmware info box also states no Databases installed.

nastea1, 2022-08-08:
Official firmware shut it down, "The marauder app is not a part of our firmware. You're probably using a fork, please file the issue in their repo as we have no relation to them."

nastea1, 2022-08-08:
Another reply "Do you hotplug wifi module?

3.3v ext shared with sd card, abnormal load spikes may cause DCDC to reset power, which is causing SD reset.
In general it's better to enable 5V-ext and not to hotplug wifi module."

nastea1, 2022-08-08:
While best practice may not be "hot plugging" appears it's resolved by enabling 5v on the gpio, which makes sense, now..

closed 2022-08-08 by nastea1