RogueMaster/flipperzero-firmware-wPlugins #216: Crash during syncronization by bluetooth with the Flipper App on iOS

Mareg74, 2022-08-10:
### Describe the bug.

When I try to syncronize the Flipper with the iOS app it'll crash the flipper, reboot it and show me this message

### Reproduction

1. Open iOS app
2. Connect to your Flipper
3. After starting syncronization with the app the flipper will crash and restart
[image]

### Target

0.63.5-0809-RM-V3

### Anything else?

Tested on an other new Flipper and other phone (my brother's one) still the same problem

RogueMaster, 2022-08-10:
https://github.com/flipperdevices/flipperzero-firmware/issues/1570

closed 2022-08-10 by RogueMaster

galactricks, 2022-08-10:
Have the same issue on android