RogueMaster/flipperzero-firmware-wPlugins #218: Crash on file download

DrBu7cher, 2022-08-10:
### Describe the bug.

When I try to download a file with qFlipper, the flipper hard resets and crashes.
Also when I connect via bluetooth it gets into a crash-loop. If my phone connects via bluetooth, the flipper instantly crashes with 'HardFault'.

### Reproduction

1. Factory Reset Flipper
2. Update to latest official 'v0.63.3'
3. Create folder in 'update' on SD-Card
4. Copy files from release-zip into that
5. Run update via qFlipper screen share -> file browser on flipper
6. Try to download e.g. 'SD Card/subghz/assets/' via qFlipper file browser
7. Watch crash. (Sometimes nothing happened at all until I pressed a key)

### Target

0.63.5-0809-RM-V3

DrBu7cher, 2022-08-10:
The crashing of the bluetooth was also in this version, but I reset my flipper completely (deleted everything from internal and SD) and did the same steps from above, now I can connect via bluetooth without problems.

RogueMaster, 2022-08-11:
This should be fixed on this next release