RogueMaster/flipperzero-firmware-wPlugins #220: Sub-Ghz - Lock keyboard do not work

miha-perfecto, 2022-08-11:
### Describe the bug.

Sub-Ghz - Lock keyboard do not work

### Reproduction

1. Go to Sub-Ghz
2. Go to Read
3. Go to Config
4. Press "Lock Keyboard"

RogueMaster, 2022-08-12:
Please see if this problem occurs in the 0812 v3 version

Foul, 2022-08-12:
Still not working in 0.64.1-0812-RMv3

RogueMaster, 2022-08-12:
Checking into why it may not work... Just confirmed all of https://github.com/flipperdevices/flipperzero-firmware/pull/1343 is in there

RogueMaster, 2022-08-13:
@PolymerPrints it seems this commit https://github.com/RogueMaster/flipperzero-firmware-wPlugins/releases/tag/0.63.5-0806-RM-NOFAP broke the Lock Keyboard. Could you review how to get it working again? Thank you!

PolymerPrints, 2022-08-13:
@RogueMaster Sure, I'll have a look

RogueMaster, 2022-08-13:
Fixed on latest commit by @PolymerPrints , thank you!

closed 2022-08-13 by RogueMaster