RogueMaster/flipperzero-firmware-wPlugins #223: SubGHz us broken again

port513, 2022-08-12:
### Describe the bug.

Have extended range activated and ignore default TX region activated.
For the last three or so FW this file does not work, claims that range is outside default range, why is it outside default range?

https://raw.githubusercontent.com/UberGuidoZ/Flipper/945362d3a8058f082abd7e6dbe57864d95d363f8/Sub-GHz/Restaurant_Pagers/LRS_Pagers/BruteForceRest.sub

### Reproduction

1. Switch on
2. Go to saved SubGHz
3. Try to open the file mentioned in description
4. Even with extended range enabled and all frequencies added it won't run

### Target

Latest RougeMaster FW

### Anything else?

Reported before and it got a fix but is now back again

RogueMaster, 2022-08-12:
Try the latest v3

closed 2022-08-12 by RogueMaster

port513, 2022-08-12:
Fixed it, really quick fix 😁