RogueMaster/flipperzero-firmware-wPlugins #233: Custom Name not changing Bluetooth MAC address

BoBeR182, 2022-08-16:
### Describe the bug.

Running `CUSTOM_FLIPPER_NAME=Name ./fbt COMPACT=1 DEBUG=0 updater_package` and installing the package is successful.
The flipper is renamed and the serial number is changed correctly.
The Bluetooth of the mac address does not match the new spoofed serial number. 

### Reproduction

1.`CUSTOM_FLIPPER_NAME=Name ./fbt COMPACT=1 DEBUG=0 updater_package`
2. Flash firmware
3. Check bluetooth

RogueMaster, 2022-08-16:
This has been changed as much as it's going to be. If you want to test and PR some code that would fix your issue, feel free

closed 2022-08-16 by RogueMaster

BoBeR182, 2022-08-16:
Documentation shows different expected behavior.
https://github.com/RogueMaster/flipperzero-firmware-wPlugins/blob/unleashed/documentation/CustomFlipperName.md?plain=1#L13
Should I open a bug upstream?

BoBeR182, 2022-08-17:
I don't understand what the issue is. I opened a bug, it got closed as a won't do. I fixed the bug with code from upstream and the pull request was closed. Is there a reason you do not want the MAC address of the BT to be changed to protect user privacy?

RogueMaster, 2022-08-17:
You just copied the unleashed file and tried to push it here without knowing anything of what you are changing?

You did not fix anything. Your changes would break the Bluetooth rename. Your changes claim to do something that is already in the code, go read it