RogueMaster/flipperzero-firmware-wPlugins #235: Mobile App unable to Connect with latest release

thespawnat, 2022-08-16:
### Describe the bug.

installed today morning the latest FW (0.64.5-0815-RMv2) and noticed, that the Android Mobile App isn't connecting to the Flipper properly.

It shows in the top right corner "Rogue FlipperName" and 2 "problematic icons" which can not be displayed.

At the bottom on the left-handside it Shows "Disconnect" next to the Bluetooth icon, but nothing from the Flipper itself is displayed like FW-Version, Build Date etc. Only see the "syncing button" spinning wild without any progress

I pressed "disconnect flipper" in the app and on the flipper "Forget all paired devices". when I try to re-pair again, the mobile app is unable to find the flipper-device and shows a retry button in the flipper-mobile-app.

when I open the bluetooth devices from mobile itself, I see the Flipper Device as "visible", but next to the name are the 2 problematic icons displayed and it's unable to pair.

Did some testing and with 0.63.5-0808-RM-V3 is the name correct again and I'm able to pair the App properly and connect.
then I installed the 0.64.2-0813-RMv2 and the connection to the mobile app worked. Main-Difference in the mobile app is the Name of the Flipper with "Rogue FLIPPERNAME 2 problematic Icons", but everything else works.

then flashed the latest FW again and the app stops working

### Reproduction

install latest FW
open mobile app -> not able to connect
install FW "0.63.5-0808-RM-V3" or "0.64.2-0813-RMv2" and connection works

### Target

0.64.5-0815-RMv2

thespawnat, 2022-08-16:
[image]

PolymerPrints, 2022-08-16:
Same issue is occurring on iOS.

kaosproject, 2022-08-16:
I just disconet the Flipper on both side and reconected and  now it's working fine

non-bin, 2022-08-16:
Issue is still present. (Android)
I turned off Bluetooth on the flipper, pressed 'forget all devices', removed the flipper from my phone, and reinstalled the app.

PolymerPrints, 2022-08-16:
> I just disconet the Flipper on both side and reconected and now it's working fine

For me it works again after re-pairing (on iOS).

non-bin, 2022-08-16:
The connection implementation (for android at least) contains this line, which will ignore any device whose name doesn't start with `Flipper`. I'll make a pull request to add a debug option to change the value (in RogueMaster and the app) unless there's a better suggestion

```kotlin
        ).filter { it.device.name?.startsWith(Constants.DEVICENAME_PREFIX) == true }
```
[components/bridge/impl/src/main/java/com/flipperdevices/bridge/impl/scanner/FlipperScannerImpl.kt](https://github.com/flipperdevices/Flipper-Android-App/blob/dev/components/bridge/impl/src/main/java/com/flipperdevices/bridge/impl/scanner/FlipperScannerImpl.kt)

RogueMaster, 2022-08-16:
it works on ios, but not on android... code already was pushed to fix this so next release will work on android

non-bin, 2022-08-16:
was it a change to the RogueMaster firmware or the app?

RogueMaster, 2022-08-16:
my firmware, I changed the Bluetooth name from broadcasting "Flipper "+name to "zzRogue "+name. App works fully in IOS but doesn't work for Android I guess. Android detection should ideally allow for a name change.

RogueMaster, 2022-08-16:
I made https://github.com/flipperdevices/Flipper-Android-App/issues/335

RogueMaster, 2022-08-16:
Re-opened with better name 😄

thespawnat, 2022-08-17:
flashed latest FW (0.64.5-0817-RM) and with "original" Flipper Device Name is the Android App working again

RogueMaster, 2022-08-17:
They don't want to fix it because they are too lazy to do it right. I guess our flippers will have to call themselves out until they correct their humongous error. Latest release fixes the issue.

closed 2022-08-17 by RogueMaster

Foul, 2022-08-30:
I don't know if it's related, but at one point I had a "remote Flipper" in bluetooh devices, which I could connect to use keyboards/mouse/media/jiggler etc in plugins, it doesn't I haven't had any more since a few updates.. and therefore all these plugins no longer work...

dakhnod, 2022-09-09:
@RogueMaster sorry to bother you, but I don't see how it's fixed. [The code](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/blob/420/firmware/targets/f7/furi_hal/furi_hal_version.c#L95) still seems to use zzRogue, doesn't it?

RogueMaster, 2022-09-09:
They supposedly updated the mobile app to detect flipper differently to allow for this rename