RogueMaster/flipperzero-firmware-wPlugins #241: Transmission is blocked at my region

Mika73, 2022-08-17:
### Describe the bug.

Transmission is blocked at my region. 

### Reproduction

When trying to send Megacode from Misc directory get error that transmission is blocked at my region. 
[image]

### Target

Transmission is blocked at your region

port513, 2022-08-17:
Have you updated extended_range.txt and setting_user.txt to support extended ranges?

You find them in subghz/assets on you SD card

Mika73, 2022-08-17:
I have copy old ones from computer to sd card and replace those files there. Maybe something wrong with files I have. Is there link to working ones so can try replace those?

port513, 2022-08-17:
Can upload mines, just a notice that the extended_range.txt is now named extended_range.txt.

You can find settings here that I just drop in my assets folder after a FW flash

https://ucarecdn.com/a508410b-97fe-4c6a-9ee7-d26df864e4b0/

RogueMaster, 2022-08-17:
Install latest release, update `extend_range.txt`, try locked file such as TPMS sub @ 300MHz

closed 2022-08-17 by RogueMaster