RogueMaster/flipperzero-firmware-wPlugins #242: Flipper crashes when pressing the left button

sundset, 2022-08-17:
### Describe the bug.

When pressing the left button (start screen) and the clock appears, flipper does a hard crash and reboots. 

### Reproduction

1. Switch Flipper on
2. Press < button to open Clock
3. Flipper crashes and reboots

zviratko, 2022-08-17:
Can confirm. Opening Clock for the first time worked, it showed the time, refreshed correctly and I was able to quit it cleanly. When opening it again, it crashed instantly. Now it crashes every time.

PolymerPrints, 2022-08-17:
Clock is broken in latest release (@RogueMaster is aware).

RogueMaster, 2022-08-17:
If you can, test with current code

pretzelmonies, 2022-08-18:
Also can confirm with newest release 0.64.5-0817-RM. Crash occurs from left key press from home screen, or "clock" app from main menu.

RogueMaster, 2022-08-18:
Fixed on todays release

closed 2022-08-18 by RogueMaster