RogueMaster/flipperzero-firmware-wPlugins #263: Flipper crashes when selecting "Name Changer" in Plugins

Painkiller8818, 2022-08-22:
### Describe the bug.

Just try to start Name Change in Plugins and flipper is crashing.

### Reproduction

1. goto Plugins
2. select "Name Changer"
3. Enter
4. Device crashes

RogueMaster, 2022-08-22:
Yup, this feature is experimental but works most of the time

sschueller, 2022-08-23:
On my device it will crash every time no exceptions. I tried it 10 times in a row. Running ```0.64.6-0823-RM420``` which replaced ```Flipper Zero Unleashed Firmware``` via qFlipper

joebetsill, 2022-08-24:
> On my device it will crash every time no exceptions. I tried it 10 times in a row. Running `0.64.6-0823-RM420` which replaced `Flipper Zero Unleashed Firmware` via qFlipper

Do you have a file named `name.txt` in the root of your SD card? Mine was crashing until I added this file. I didn't know what to put in it so I just typed the name I wanted but then the plugin asked me to type the new name, so I suspect the `name.txt` file can be blank.

Painkiller8818, 2022-08-24:
> Do you have a file named `name.txt` in the root of your SD card?

This file should not be in the root, it is in the Dolphin folder, there should be this file and you can edit it, it is a description in the file how to do and whats the limitations.

joebetsill, 2022-08-24:
> > Do you have a file named `name.txt` in the root of your SD card?
> 
> This file should not be in the root, it is in the Dolphin folder, there should be this file and you can edit it, it is a description in the file how to do and whats the limitations.

Ah yeah, that's right. I'll edit. 

Where do you find the "real" file with instructions? It definitely wasn't there after flashing the firmware.

Painkiller8818, 2022-08-24:
> Where do you find the "real" file with instructions? It definitely wasn't there after flashing the firmware.

It was there after an update for me but if its not for you this is the original content:

Filetype: Flipper Name File
Version: 1
# Changing the value below will change your FlipperZero device name.
# Note: This is limited to 8 characters using the following: a-z, A-Z, 0-9, and _
# It can contain other characters but use at your own risk.
Name: YOURNAME

Painkiller8818, 2022-08-24:
sry for the big font, no idea why it makes this big

joebetsill, 2022-08-24:
I decided to pull my sd card and it was probably just coincidence, but I did, in fact, place a `name.txt` file in the root, and the rename plugin started working immediately after, where before it was crashing every time I tried to use the plugin. 

Looking under `/dolphin' I have the appropriate `name.txt` file, formatted as you say.

So, I don't know what's going on, but it's almost certainly just a coincidence that adding that file was followed by the plugin working. 

(\# denotes H1 formatting in markdown; \## denotes H2, \### denotes H3.)

Painkiller8818, 2022-08-24:
glad i could help :)

Painkiller8818, 2022-08-24:
Flipper crashes because of name.txt file having a wrong format or windows CR spacing instead of a linux Spacing.

After opening with a linux file editor or on windows changing eg. notepad++ to change to linux spacing and save it worked

closed 2022-08-24 by Painkiller8818

Foul, 2022-08-24:
> Flipper crashes because of name.txt file having a wrong format or windows CR spacing instead of a linux Spacing.
> 
> After opening with a linux file editor or on windows changing eg. notepad++ to change to linux spacing and save it worked

Confirming : was crashing/rebooting too... saved name.txt on linux, edit with kate, saved, re upload on flipper.. and it works :)