RogueMaster/flipperzero-firmware-wPlugins #267: FlipFrid build error when enabled in the applications/meta/application.fam

konstcranky, 2022-08-24:
### Describe the bug.

FlipFrid build errors due to lfrfid changes in the 65f6f6843d8185bc98ec45829230f14fc184589e

### Reproduction

diff --git a/applications/meta/application.fam b/applications/meta/application.fam
index 55836120..dd6c7428 100644
--- a/applications/meta/application.fam
+++ b/applications/meta/application.fam
@@ -49,7 +49,7 @@ App(
         # "APPS_SentrySafe",
         "APPS_SpectrumAnalyzer",
         "APPS_WAVPlayer",
-        # "APPS_FlipFrid",
+        "APPS_FlipFrid",
     ],
 )

$ ./fbt
        APPS    build/f7-firmware-C/applications/applications.c
        CPP     applications/FlipFrid/flipfrid.cpp
cc1plus: warning: command-line option '-Wimplicit-function-declaration' is valid for C/ObjC but not for C++
        CPP     applications/FlipFrid/flipfrid_app_launcher.cpp
cc1plus: warning: command-line option '-Wimplicit-function-declaration' is valid for C/ObjC but not for C++
        CC      applications/VideoPoker/poker.c
applications/FlipFrid/flipfrid.cpp:6:10: fatal error: ../lfrfid/helpers/rfid_timer_emulator.h: No such file or directory
    6 | #include "../lfrfid/helpers/rfid_timer_emulator.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
        CC      applications/about/about.c
        CC      applications/archive/helpers/archive_apps.c
        CC      applications/archive/helpers/archive_browser.c
        CC      applications/archive/helpers/archive_favorites.c
        CC      applications/archive/helpers/archive_files.c
        CC      applications/archive/scenes/archive_scene.c
        CC      applications/archive/scenes/archive_scene_browser.c
        CC      applications/archive/scenes/archive_scene_delete.c
        CC      applications/archive/scenes/archive_scene_rename.c
        CC      applications/archive/views/archive_browser_view.c
        CC      applications/archive/archive.c
        CC      applications/arkanoid/arkanoid_game.c
scons: *** [build/f7-firmware-C/applications/FlipFrid/flipfrid.o] Error 1
********** ERRORS **********
Failed building build/f7-firmware-C/applications/FlipFrid/flipfrid.o: Error 1

Ganapati, 2022-08-24:
Fixed with PR https://github.com/RogueMaster/flipperzero-firmware-wPlugins/pull/268
flipfrid will now use the new LFRFID api.

RogueMaster, 2022-08-24:
it's back in

closed 2022-08-24 by RogueMaster