RogueMaster/flipperzero-firmware-wPlugins #271: Internal storage is only 32 kb

fraski30, 2022-08-24:
### Describe the bug.

After clean install rm 0.65.1-823RMv2 with micro sd inserted internal total storage is only 32kb. it´s impossible to install/update new firmware without do complete restore with qiflash. Because flipper says need more free space in internal storage.
With OFW internal total space it´s about 100 kb, with last rm only 32 kb
 

### Reproduction

try to flash or update rm or any other firmware. Flipper says need more internal space.

RogueMaster, 2022-08-24:
RM uses more space, yes, this is not something we need an issue report for

closed 2022-08-24 by RogueMaster

fraski30, 2022-08-24:
Thanks, so every time I have to update rm I have to format flipper?

RogueMaster, 2022-08-24:
No, only on a few releases till FAPs are done

fraski30, 2022-08-24:
Thank you for you work!