RogueMaster/flipperzero-firmware-wPlugins #275: U2F

RogueMaster, 2022-08-26:
### Describe the enhancement you're suggesting.

Who uses U2F? I could disable it for 15+KB and re-add it when faps return...

### Anything else?

I don't use it, I prefer more games...

closed 2022-08-26 by RogueMaster

Foul, 2022-08-26:
i'm using it, Security keys on Github :D

Painkiller8818, 2022-08-26:
i also use it for google, fb etc. almost all stuff where i can use it, except microsoft :D

Gallocon, 2022-09-02:
I use it. Games are fun for experimentation and pushing boundaries, but I've yet to find a game on Flipper worth killing U2F over.

I cannot wait until the FAP loader is working so you don't have to do this song-and-dance with features. I can tell it's been a difficult battle.

fobrice, 2022-09-02:
Like @Foul  & @Painkiller8818 use it for google, fb, github, private gitlab and linux auth with pam-u2f.

t0mab, 2022-09-02:
Hi me as well beginning to use it when I received the flipper and I think it's worth keeping !
Same as @Foul , @Painkiller8818, @fobrice mainly for github, private gitlab, and linux pam (sudo without password & more)