RogueMaster/flipperzero-firmware-wPlugins #284: Error - Need more free space in internal storage on Release 0.65.3-0829-RM.Av2

PifPof73, 2022-08-30:
### Describe the bug.

Hi, 
first - thanks for your great work!
I reguarly update my flipper-zero with your firmware.
Now you mention a two step update (side B), and I try to update like used (calling the updater from browser), but get an error.

"Error - Need more free space in internal storage"

Could you please adjust the documentation how to work with the twoside Update?

### Reproduction

1. Uzip release and copy it in update folder.
2. Browse to update and call update in app
3. Receive "Error - Need more free space in internal storage"

### Anything else?

Regards,
PifPof73

semchapeu, 2022-08-30:
If I understand it correctly, this will be an issue until the FAP Loader is implemented.

Foul, 2022-08-30:
Save you INTERNAL storage
FLASH
Recover INTERNAL storage
Reboot
Enjoy :)

PifPof73, 2022-08-30:
> Save you INTERNAL storage FLASH Recover INTERNAL storage Reboot Enjoy :)

you mean: 

1. Plugins -> Dolphin Backup (save INTERNAL storage)
2. Settings -> Storage -> Factory Reset
3. Browser -> Update -> Update (FLASH)
4. Reboot (Automatically)
5. Plugins -> Dolphin Restore
6. Enjoy :)

???

Foul, 2022-08-30:
You can copy Internal/restore Files by hand with qFlipper too...

closed 2022-08-30 by RogueMaster