RogueMaster/flipperzero-firmware-wPlugins #285: Prob compiling full Firmware

Foul, 2022-08-30:
### Describe the bug.

when compiling my own firmware, i'm using like doc said :
./fbt flash_usb_full

With **Original FW**, no **prob Compiling**.

With Rogue :

./fbt flash_usb_full
        CDB     build/f7-firmware-C/compile_commands.json
Setting build/f7-firmware-C as latest built dir (./build/latest/)
scons: *** [assets/resources/Manifest] TypeError : Tried to lookup Dir 'assets/resources/update/0.65.3-0829-RM-SideB' as a File.
********** ERRORS **********
Failed building assets/resources/Manifest: TypeError : Tried to lookup Dir 'assets/resources/update/0.65.3-0829-RM-SideB' as a File.

Seems to be here since A/B Package. Before.. was compiling fine.

### Reproduction

git clone --recursive https://github.com/RogueMaster/flipperzero-firmware-wPlugins.git
cd flipperzero-firmware-wPlugins/
./fbt flash_usb_full

RogueMaster, 2022-08-31:
Remove dist folder, build folder and the sconsdb file and try again.

closed 2022-08-31 by RogueMaster

Foul, 2022-08-31:
no sconsdb file present..
Deleting dist & build seem to do the trick  👍