RogueMaster/flipperzero-firmware-wPlugins #288: add on main firmware the: flipperzero-sentry-safe-plugin

MrCook1es, 2022-08-31:
### Description of the feature you're suggesting.

Hi, athanks for your great work!
I reguarly update my flipper-zero with your firmware.

It will be great if you implemented the **flipperzero-sentry-safe-plugin** on the RM release .
https://github.com/H4ckd4ddy/flipperzero-sentry-safe-plugin

congratulations again for your work

Foul, 2022-08-31:
Already included in B side :

Latest Updates:
Apps HIDDEN for compile space can be found on Side B in the update folder.

MrCook1es, 2022-08-31:
I'm sorry but I can't find "Side B" or the "update folder" :'(

RogueMaster, 2022-08-31:
Use SD install method for Side A

closed 2022-08-31 by RogueMaster

MrCook1es, 2022-08-31:
> 

Thanks a lot for the reply!
sorry, I looked in the firmware 0.65.3-0825-RM and there the update folder is empty.
This is a change that you only made in the .Av2
Thanks and great job!