RogueMaster/flipperzero-firmware-wPlugins #294: No iButton in latest release(0.65.4-0901-RM.Av2)

Bosek, 2022-09-01:
### Describe the bug.

After flashing latest release the iButton menu option disappeared. Tried to flash the latest official fw, then the one from this repo. Latest Unleashed release from two days ago works fine.

### Reproduction

1. Flash latest official FW via qFlipper
2. Flash 0.65.4-0901-RM.Av2 release via qFlipper
3. Try to find iButton in menu

Foul, 2022-09-01:
Flash B side !!! Read The doc.

gustavo8115, 2022-09-02:
Why don't they include the ibutton function in version A ? Since it is a main function of flipperZero. Version B doesn't have U2F and BADUSB.

RogueMaster, 2022-09-02:
> Why don't they include the ibutton function in version A ? Since it is a main function of flipperZero. Version B doesn't have U2F and BADUSB.

Why don't you compile your own?

closed 2022-09-02 by RogueMaster