RogueMaster/flipperzero-firmware-wPlugins #298: The Passport on realease 0.65.4-0901-RM.Av2 is hard crashing flipper

chazp246, 2022-09-02:
### Describe the bug.

Flipper will crash if i try to open passport. Both by right button and also in the settings.

### Reproduction

1a. press right button on the unlocked home screen
1b. open passport from within the setings

RogueMaster, 2022-09-02:
Will review for all levels

toxyl, 2022-09-06:
On my device the name I gave it was reset on reboot. I was able to set it again using the Name Changer, but any subsequent attempts to use the Name Changer resulted in Hard Faults. Even after reboots the Name Changer kept crashing, but after opening the Passport once, I don't experience any crashes with the Name Changer. Once I update the name, the above starts anew. I'm also using 0.65.4, maybe these two issues are related?

RogueMaster, 2022-09-07:
Fixed on V2

closed 2022-09-07 by RogueMaster