RogueMaster/flipperzero-firmware-wPlugins #314: Accessing Subghz shows "Firmware Update Needed"

elgumso, 2022-09-11:
### Describe the bug.

Following installation of the latest release update (0.66.2-0911-RM.Av2), accessing the Subghz menu shows a "Firmware update needed" notification and Subghz functionality cannot be used. Same issue when accessing a sub file from the file explorer. The Subghz Bruteforcer does appear to have no problem though.

### Reproduction

Access subghz menu, notification occurs stating update needed preventing access to subghz functionality.

### Target

0.66.2-0911-RM.Av2

### Anything else?

Have tried SD card update, DFU flash, etc. all with same results.

davetarget, 2022-09-11:
I have the same issue

duan78, 2022-09-11:
Same problem here

RogueMaster, 2022-09-11:
Please confirm this fixes it... 
[0.66.2-0911-RM.Av3.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/9542721/0.66.2-0911-RM.Av3.zip)

Sam-Dowling, 2022-09-11:
I bypassed this issue after a few failed attempts by doing the following:

- Completely formatting the sdcard
- Flashing to stock through qflipper
- Updating to 0.66.2-0911-RM.Av2 via SD card

> Please confirm this fixes it... [0.66.2-0911-RM.Av3.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/9542721/0.66.2-0911-RM.Av3.zip)

This worked for me without needing to do above steps like before.

duan78, 2022-09-11:
> Please confirm this fixes it... [0.66.2-0911-RM.Av3.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/9542721/0.66.2-0911-RM.Av3.zip)

It work now ;) Thank you !

closed 2022-09-11 by RogueMaster

thespawnat, 2022-09-11:
v3 fixed the issue on my flipper