RogueMaster/flipperzero-firmware-wPlugins #318: BT Device Name missing "zzRogue" after 0.66.2-0913-RM.A

thespawnat, 2022-09-13:
### Describe the bug.

after installing 0.66.2-0913-RM.A, Android App is unable to connect. 

in the BT Devices on Android is the Flipper shown without "zzRogue".

FW 0.66.2-0911-RM.A has still "zzRogue" in the Device Name present

If you disconnect / forget the Device, you can reconnect it fine with the "new" name

### Reproduction

install 0.66.2-0913-RM.A
scan BT Devices -> Flipper without "zzRogue" Naming
restore 0.66.2-0911-RM.A
scan BT Devices -> Flipper with "zzRogue" Naming

### Target

0.66.2-0913-RM.A

Antti, 2022-09-13:
It's the other way round for me. With zzRogue android app couldn't find my flipper.

Foul, 2022-09-13:
For me it's better.. i see multiple flipper lol
[image]

RogueMaster, 2022-09-13:
this was changed back in last release to behave just like unleashed / stock 'xFlipper'

closed 2022-09-13 by RogueMaster